stockmann-lab/shim_amp_hardware_linear #2: Create connector board for low-density (7 x 6) ODU contacts

dcstraney, 2023-01-28:
16 channels per connector: 2 ribbon cables/connector
Needs pinout guidance from @jaystock

jaystock, 2023-02-02:
Jason can repin 5 twisted pairs on the AC/DC coil.  Remove 5 pairs from Connector III and move them to Connector II
Pin Connector I from 1A down to 3C (insulating bodies 1-3), leave last 5 pairs unconnected
Pin Connectors II and III up from the bottom, leave the top five pairs unconnected.  
So use III: 7H → 9K and II: 4H → 6K

Key: 
   Roman numeral refers to the cable number (I, II, III) 
  Insulating body is 1-9 
  For each insulating body, the pin pairs are  A, C, D, E, G, H, K (7 pairs)
  14 pints per insulating body, 42 pins per cable

dcstraney, 2023-02-02:
2x layouts:
All pairs are horizontal
Connector 1: Fill all pairs starting from beginning (labeled "A" on insulating body; top-left? bottom-right? facing male or female?) towards end; last 5 pairs will be unused
Connectors 2, 3: Fill all pairs starting from end, towards beginning; first 5 pairs will be unused

jaystock, 2023-02-02:
[image]

jaystock, 2023-02-02:
Here is a photo showing the ordering of insulating bodies on each ODU connector.  They are going left-to-right when viewed from the outside of the enclosure

jaystock, 2023-02-02:
<google-sheets-html-origin><style type="text/css"></style>

  | Unique identifier |   |   |   |   | Channel address | Roman numeral |   |   | Channel address | RF element mapping
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
  | Label (plug / board) | Done? | Board | Channel | CH output | by teensy | ODU Plug number | INSULATING BODY | ODU pin pair | by teensy (0-ind) | 46ch coil channel assignment
Top Half | 1A / 1 | x | 3 | 1 | 1 | 0 | I | 1 | A | 0 | 13
1C / 2 | x | 3 | 2 | 2 | 1 | I | 1 | C | 1 | 18/19
1D / 3 | x | 3 | 3 | 3 | 2 | I | 1 | D | 2 | 11
1E / 4 | x | 3 | 4 | 4 | 3 | I | 1 | E | 3 | 7
1G / 5 | x | 3 | 5 | 5 | 4 | I | 1 | G | 4 | 10
1H / 6 | x | 3 | 6 | 6 | 5 | I | 1 | H | 5 | 6
1K / 7 | x | 3 | 7 | 7 | 6 | I | 1 | K | 6 | 2
2A / 8 | x | 3 | 8 | 8 | 7 | I | 2 | A | 7 |  
2C / 9 | x | 4 | 1 | 9 | 8 | I | 2 | C | 8 | 3
2D / 10 | x | 4 | 2 | 10 | 9 | I | 2 | D | 9 | 9
2E / 11 | x | 4 | 3 | 11 | 10 | I | 2 | E | 10 | 4
2G / 12 | x | 4 | 4 | 12 | 11 | I | 2 | G | 11 | BROKEN INSULATING BODY
2H / 13 | x | 4 | 5 | 13 | 12 | I | 2 | H | 12 | 1
2K / 14 | x | 4 | 6 | 14 | 13 | I | 2 | K | 13 | 5
3A / 15 | x | 4 | 7 | 15 | 14 | I | 3 | A | 14 | 12
3C / 16 | x | 4 | 8 | 16 | 15 | I | 3 | C | 15 | 8
Reserved for future expansion to 62ch | 3D / 17 |   |   | 1 | 17 |   | I | 3 | D |   |  
3E / 18 |   |   | 2 | 18 |   | I | 3 | E |   |  
3G / 19 |   |   | 3 | 19 |   | I | 3 | G |   |  
3H / 20 |   |   | 4 | 20 |   | I | 3 | H |   |  
3K / 21 |   |   | 5 | 21 |   | I | 3 | K |   |  
4A / 22 |   |   | 6 | 22 |   | II | 4 | A |   |  
4C / 23 |   |   | 7 | 23 |   | II | 4 | C |   |  
4D / 24 |   |   | 8 | 24 |   | II | 4 | D |   |  
4E / 25 |   |   | 1 | 25 |   | II | 4 | E |   |  
4G / 26 |   |   | 2 | 26 |   | II | 4 | G |   |  
4H / 27 |   |   | 3 | 27 |   | II | 4 | H |   |  
4K / 28 |   |   | 4 | 28 |   | II | 4 | K |   |  
5A / 29 |   |   | 5 | 29 |   | II | 5 | A |   |  
5C / 30 |   |   | 6 | 30 |   | II | 5 | C |   |  
5D / 31 |   |   | 7 | 31 |   | II | 5 | D |   |  
5E / 32 |   |   | 8 | 32 |   | II | 5 | E |   |  
Bottom half | 5G / 33 | x | 5 | 1 | 33 | 16 | II | 5 | G | 16 | 10b
5H / 34 | x | 5 | 2 | 34 | 17 | II | 5 | H | 17 | 17b
5K / 35 | x | 5 | 3 | 35 | 18 | II | 5 | K | 18 | 16b
6A / 36 | x | 5 | 4 | 36 | 19 | II | 6 | A | 19 | 21b
6C / 37 | x | 5 | 5 | 37 | 20 | II | 6 | C | 20 | 22b
6D / 38 | x | 5 | 6 | 38 | 21 | II | 6 | D | 21 | 27b
6E / 39 | x | 5 | 7 | 39 | 22 | II | 6 | E | 22 | 29b
6G / 40 | x | 5 | 8 | 40 | 23 | II | 6 | G | 23 | 14b
6H / 41 | x | 6 | 1 | 41 | 24 | II | 6 | H | 24 | 15b
6K / 42 | x | 6 | 2 | 42 | 25 | II | 6 | K | 25 |  
7A / 43 | x | 6 | 3 | 43 | 26 | III | 7 | A | 26 | 1b
7C / 44 | x | 6 | 4 | 44 | 27 | III | 7 | C | 27 | 2b
7D / 45 | x | 6 | 5 | 45 | 28 | III | 7 | D | 28 | 6b
7E / 46 | x | 6 | 6 | 46 | 29 | III | 7 | E | 29 | 12b
7G / 47 | x | 6 | 7 | 47 | 30 | III | 7 | G | 30 | 11b
7H / 48 | x | 6 | 8 | 48 | 31 | III | 7 | H | 31 | 24b
7K / 49 | x | 7 | 1 | 49 | 32 | III | 7 | K | 32 | 5b
8A / 50 | x | 7 | 2 | 50 | 33 | III | 8 | A | 33 | 13b
8C / 51 | x | 7 | 3 | 51 | 34 | III | 8 | C | 34 | 3b
8D / 52 | x | 7 | 4 | 52 | 35 | III | 8 | D | 35 | 7b
8E / 53 | x | 7 | 5 | 53 | 36 | III | 8 | E | 36 | 9b
8G / 54 | x | 7 | 6 | 54 | 37 | III | 8 | G | 37 | 8b
8H / 55 | x | 7 | 7 | 55 | 38 | III | 8 | H | 38 | 29b
8K / 56 | x | 7 | 8 | 56 | 39 | III | 8 | K | 39 | 18b
9A / 57 | x | 8 | 1 | 57 | 40 | III | 9 | A | 40 | 25b
9C / 58 | x | 8 | 2 | 58 | 41 | III | 9 | C | 41 | 30b
9D / 59 | x | 8 | 3 | 59 | 42 | III | 9 | D | 42 | 31b
9E / 60 | x | 8 | 4 | 60 | 43 | III | 9 | E | 43 | 20b
9G / 61 | x | 8 | 5 | 61 | 44 | III | 9 | G | 44 | 4b
9H / 62 | x | 8 | 6 | 62 | 45 | III | 9 | H | 45 | 19b
9K / 63 | x | 8 | 7 | 63 | 46 | III | 9 | K | 46 | 26b
  | GOES TO 4A!!! |   | 8 | 8 | 64 | 47 | spare |   |   | 47 |

dcstraney, 2023-05-17:
Created first-pass boards as of [efbc150](https://github.com/stockmann-lab/shim_amp_hardware_linear/commit/efbc150b45d7eb25cd1035c0515346c25bce584f)