stockmann-lab/shim_amp_hardware_linear #23: Power-up with Snickerdoodle backplane has 0V setpoints

dcstraney, 2025-01-05:
When powered up with dummy loads connected to amp channels, Lincoln's hardware setup (D4 amp control board with ADC/DAC reset rework #14 + add'l 10µF on ADC/DAC reset) triggers power supply current limit.

From Lincoln, probed @ amp control board:
Ch1: DAC ch0 voltage
Ch2: LDACn
Ch3: SCK
Ch4: +5V
Triggered off +5V = 2V rising edge
[image]
[image]
[image]

Hash @ ~1.5 sec is likely Snickerdoodle startup, and FPGA configuration process.
Power stage is supposed to power up @ ~300 ms.
ADC & DAC should come out of reset @ ~100 ms (10KΩ * 10µF); the DAC ch0 transition @ ~80 ms is probably due to this.

Open questions:
- DAC is supposed to output the reset level (2.048V) on reset, but in this case it looks like it goes to the correct level only on coming _out of_ reset.  Why?  DAC reset, acc. to datasheet, may be edge-sensitive and _not_ level-sensitive, which means that it may need a two-stage reset rather than a simple RC circuit here.  In practice, this shouldn't cause a problem because power stage only is enabled *later*, but still relevant.
- When does DAC ch0 output go to 0V? (which causes the high output currents)  Seems to be ok through the first couple seconds, but Lincoln's measurements show it at 0V when measured with a meter after startup.  Alternatively, why does the power stage draw excessive current if ch0 is 2V? (are some of the other channels still at 0V, esp. if reset doesn't happen correctly?)
- Why does DAC ch0 output go positive for ~5 ms before the +5V comes up?

dcstraney, 2025-01-06:
Attempting to reproduce with Montreal setup:
DAC-reset rework needs repair(?) as RC-delay output is shorted to gnd, but shows DAC output going to 2V correctly (@ 40 ms), and then later dropping to 0V (@ 170 ms).
[image]

dcstraney, 2025-01-06:
+3.3V rail comes up at the same time as +5V:
[image]

dcstraney, 2025-01-06:
Definitely related to power-up spurious signals on comms lines from Snickerdoodle: powering up with 8p8c Ethernet cables disconnected (no DAC communications to amp control board) results in 2.048V DC on all DAC outputs.
[image]

dcstraney, 2025-01-08:
DAC channels all now stay at 2.048V when LDAC+, DAC_CS+, DAC_MOSI+, and SCKI+ are all pulled up to +3.3V, and LDAC-, DAC_CS-, DAC_MOSI-, and SCKI- are all pulled down to gnd, with 33KΩ resistors.
[image]

dcstraney, 2025-02-07:
Same fix works on Lincoln's copy; confirmed correct fix.

dcstraney, 2025-03-17:
@LincolnCB Reworks to implement Shutdown_Force -> DAC reset

Backplane (amp board 0 only)
1. Connect a wire from the bottom pad of R64, wrapping around to the bottom side of the board, and to pin 3 of "Jack B" (J6):
[image]

Amp board
(Note: easiest to use through-hole parts here, inc. a TO-92 2N3904 or similar for the transistor)
1. Connect one side of a 10-20K resistor to pin 3 of J17 / jack B:
[image]
2. Connect the other side of the resistor to the base lead of an NPN transistor
3. Connect the collector of the NPN transistor to the non-grounded side of the reworked DAC reset circuit, preferably through a thin wire to avoid putting mechanical stress on the solder joint:
[image]
4. Connect the emitter of the NPN transistor to ground.  An easy ground location to find is the exposed, gold-plated optional-shield-connection pads around the ADC & DAC:
[image]

LincolnCB, 2025-03-24:
Added pre-latch Shutdown_Force signal on Jack B with transistor on DAC control board to pull the reset signal low. The Shutdown_Force and _Reset work well, releasing the DAC reset during software operation. The DAC inputs seem to float while the signal is low.

Unfortunately, the Shutdown_Force is only asserted after the FPGA has been configured. A pull-up to Vlogic (2.5V) on the bank improves this, but the +2.5V net only settles in just after the FPGA starts to boot, meaning the DACs can hit the zero state for a short time, enough to drive the power state to a higher current draw and mess up the Snickerdoodle's power supply.

[image]

Ch1 is the +2.5V pin, Ch2 is the DAC, Ch3 is LDACn, Ch4 is the DAC reset line

dcstraney, 2025-03-27:
Alternate more-robust rework to try:
1. Cut the Shutdown_Force trace on the top side of the board:
[image]
2. Connect a wire to the same connector pin, on the bottom side of the board (9th from the top):
[image]
3. Connect that wire through a 10K resistor to the base of a 2N3904 (or similar) NPN transistor.
4. Connect the NPN's emitter to ground.
5. Connect the NPN's collector through a 15-22K resistor to +VDC.
6. Connect the NPN's collector also to the via side of R64 (or equivalent resistor on another channel).
[image]

commit 45b52282 (2025-02-08): Comms fix, misc. improvements

Added 33KΩ pull-ups & pull-downs to amp ctrl board's LVDS inputs, to solve issues caused by unconfigured FPGA pins during boot (fixes #23)
Swapped backplane power input polarity for consistency with power connectors (fixes #20)
Added "manual enable" optical input to backplane, for safety: requires positive enable to keep amplifiers enabled
commit 30cce902 (2025-04-27): Corrected power-up behavior

Backplane: Inverted polarity of Shutdown_Force (now Power_Enable_FPGA), to keep amps powered off until software starts: fixes #23
Backplane: Routed Shutdown_Force to unused pin 3 on RJ45 jack B to amp boards: fixes #23
Backplane: Added pull-up and pull-down resistors to fanout buffer inputs, to keep SCK & ~LDAC signals in proper state during power-on: fixes #23
Amp control: Added transistor to reset DAC & ADC from pin 3 of RJ45 jack B
closed 2025-04-30 by dcstraney