stockmann-lab/shim_amp_hardware_linear #25: Provision for maintaining stability when user "hot plugs" a load

jaystock, 2025-03-04:
Explore provisions for maintaining output current stability when a user "hot plugs" a load while amplifiers are powered on.   While we can advise users against doing this, in practice I think users will do this anyway either out of negligence or forgetfulness.  

On the Rev. C shim amplifiers, a 330 ohm resistor was added in parallel with the load to improve stability when hot plugging occurred.  

This is an optional feature, not essential.

dcstraney, 2025-03-06:
What do you mean by "stability" in this context?  Do you mean reducing the transient current spike that necessarily happens when the load is hot-plugged?

jaystock, 2025-03-07:
I just wanted to get a feel for what kind of behavior could occur if users hot plug their coil loads.  If the worst that can happen is a transient current spike, and not something like the power amp somehow going out of regulation, then maybe that's not a big cause for concern.  Maybe this is misplaced concern on my part.

dcstraney, 2025-03-17:
No, that's a reasonable situation to pay attention to, I just don't think "stability" is the right term here (as that has a very specific, defined meaning when it comes to control loops) and this is all premature.  Going to change the title to something more appropriate.